polyend/TrackerBetaTesting #1633: [1.6.0] freeze

modmodmodmod, 2023-02-09:
I used Tracker for the first time in a while, so I installed firmware v1.6.0 and made it.
Then suddenly all the buttons and encoders stopped working while playing song mode.
Playback sound continues.

So I shut down by pressing and holding the power button.
The shutdown waiting bar animation appeared and the power went off.
When I rebooted, the screen was in pattern mode, but all the controls didn't work and there was no sound. I rebooted again, but the state did not change.

After unplugging the SD card, backing up the data on my Mac, and running FastAid on the SD card, PT worked.

Therefore, the temporary solution was successful, but I wrote it here because I want such a bug to disappear in the future.
I submitted a project that freezes.

[042 byterat-012.zip](https://github.com/polyend/TrackerBetaTesting/files/10691603/042.byterat-012.zip)

modmodmodmod, 2023-02-09:
I ran into this problem again.
I rebooted PT several times and removed and reinserted the SD card, but nothing changed.
When I turned off the power and unplugged and plugged the power cable, the freeze was released.

PRAYTHEPO, 2023-02-09:
> I ran into this problem again.
> 
> I rebooted PT several times and removed and reinserted the SD card, but nothing changed.
> 
> When I turned off the power and unplugged and plugged the power cable, the freeze was released.

First thing to do in this situation is to unplug unit and let all capacitors to discharge, nothing unusual. Also try to change your power supply and make sure that you use electrically-stable source. Any dropouts may lead to freeze behavior. It is not SD card issue, its pure hardware.

Had quite the same problem until I changed power supply. Original one from Polyend works well, btw.

modmodmodmod, 2023-02-10:
Thank you very much. It was reassuring to know that the data was normal.
I power the original PT power supply directly from a low noise audio power supply.
But come to think of it, the USB cable used a third-party product.
I kept the original cable carefully in the warehouse, but there is crowded and I can't reach it for a while.
Anyway, I found the cause and it helped me. Thank you for your kind support.

closed 2023-04-13 by MiroPoly